UVicRocketry/Xenia1-Flight-Computer #23: X1-AV-12 HX711 research

klemie, 2022-02-10:
Find the data sheet of the HX711 in this [](https://docs.google.com/document/d/1gTJJ3SIZ1-oAkT5Fk1SbSard9b0-FxJ6s4E3crWBEo4/edit) and see if there are any combatiable python libraries if not figure out how we would read the data.

klemie, 2022-02-17:
Feb 16th update: Charlie hasn't been able to get anything done yet. Needs data sheet to look at driver info

jjderooy, 2022-02-19:
This is the datasheet. It includes a C driver that could be rewritten in python if needed. [https://www.digikey.com/htmldatasheets/production/1836471/0/0/1/hx711.html](url)

klemie, 2022-02-24:
- Meet up with @TechGal tomorrow at 4pm

klemie, 2022-03-01:
https://github.com/UVicRocketry/Xenia1-flight-Computer/issues/27#issuecomment-1054649776

klemie, 2022-03-09:
[multi HX711 Arduino](https://github.com/shmulike/HX711-multi)

klemie, 2022-03-17:
mostly the framework is done. Having Troubles changing types. Just let python choose.

Having a hard time reading reading hex from the sensor.

klemie, 2022-03-17:
@PokeyOne will help figure out c conversion

jjderooy, 2022-04-22:
Finished in #56.

closed 2022-04-22 by jjderooy