UVicRocketry/Xenia1-Flight-Computer #27: X1-AV-14 HX711 HDMI reading

klemie, 2022-02-19:
Talk to JJ and figure how we are going to read the data from his PCB to the flight computer

cjsquires, 2022-02-28:
This library works for the HX711 (and is in python) https://github.com/gandalf15/HX711/
It the python file of the driver is rather large, but should work. The file also has all the validity checks included, and is very well documented. 

If it turns out to be too much for the flight computer to handle, then we could also call the compiled c driver from python.

I can clarify anything that’s confusing to anyone (hopefully)

jjderooy, 2022-03-01:
I had a look at the python lib you linked. It looks pretty good but will need to be modified since we are  using multiple HX711s with a single clock line. Basically the way it works is the flight computer will pulse the clock line, and the HX711 will send data back on the data line. The lib you linked only allows one HX711 (ie one data line) per clock line.

Up to you whether you think it will be easier to modify the library or write your own. After reading through the code, there is a lot of extra error checking and functions which don't really apply to what we're doing like setting the output to kilograms and that sort of thing. Most of these libraries are designed to use HX711s for kitchen weigh scales which is why they have all those features.

Up to you whether you think it will be easier to modify the library or write your own. Alternatively, here is a [multi HX711 Arduino](https://github.com/shmulike/HX711-multi) library that you could convert to python.

TechGal, 2022-03-03:
Move to next sprint. 
- converting c++ to python

jjderooy, 2022-04-22:
Finished in #56.

closed 2022-04-22 by jjderooy