UVicRocketry/Xenia1-Flight-Computer #29: X1-AV-17 Write Issues for air brakes plan

klemie, 2022-02-19:
Read old repo (python airbrakes), if you are confused on anything talk to conner ebert our team lead.

klemie, 2022-03-09:
Set a date for the meeting

klemie, 2022-03-10:
Add issues to https://github.com/UVicRocketry/Xenia1-flight-Computer/milestone/3

PokeyOne, 2022-03-17:
# Mateo's Dev Notes

This comment is a work in progress; I'm just writing some notes in a github comment as I work.

Here is a diagram from the "MECH400_FinalReport_Group13.pdf" docment available on GitHub
[image]

And here is another diagram from the "software in-the-loop testing" section of the same document. Figure 28: "Simulink Model of AACS Control System and Flight Dynamics"
[image]

*I'm just going to edit this comment as I work through components that this can be broken down into*. After discussing tonight, issues can be created and branches, and then get programming.

## Components/Tasks
Here's a sorta workflow type thing on what needs to be developed
```mermaid
flowchart LR
    Issues[Create Issues] --> sensor[Sensor Data Retrieval in One Object/Struct]
    Issues --> Kalman
    Kalman --> Drag[Drag Optimizer]
    Issues --> Output[What does output look like]
    sensor --> Kalman
    Drag --> Motor
    Kalman -- error data --> Motor[Motor Controller Convert Error into steps]
    Motor --> Output
    sensor --> State["Determine when system is active [1]"]
    State --> Motor
```
[1] See "5.2.6 Active Control Flight Envelope" section of the report.

In the first diagram of this comment, they mention the "State Observer" is where the Kalman stuff would be happening.

**Here are how I'm thinking of breaking this up**:
- [ ] Transform sensor input from raw data to interpretable numbers
- [ ] Make kalman take the input data and output error data.
- [ ] Drag Optimizer (Not sure what this will involve)
- [ ] Convert drag-op and kaman values into flap angles and motor steps
- [ ] Output data back to motor and self.
- [ ] control when system active/not

All stages should include testing of that stage.

## What I'm not Sure on
- Not sure the scope of the "air brakes" plan. Seems like this is less air brakes, and more just flight controller.
- What the drag optimizer 100% does.
- How much of that report we are following exactly and how much is changed in recent development.
- The flight dynamics section of the first graph.
- How are we testing stuff.

klemie, 2022-03-17:
Discuss at meeting

PokeyOne, 2022-03-23:
## Summary of results from meeting on this issue

There is a more detailed summary of the Airbrakes meeting on the Discord under AEROSTRUCTURE in the `# | airbrakes |` channel.

I will be translating a Matlab script we have into python and we will see how well it can predict final altitude of the rocket at different points in flight. If that script works, we may just use that script instead of a full Kalman filter to determine how much we need to deploy airbrakes.

TechGal, 2022-03-24:
other predictive options: TinyML

closed 2022-04-07 by klemie