UVicRocketry/Xenia1-Flight-Computer #30: X1-AV-18 Airbakes encoder

klemie, 2022-02-19:
Not completely sure what the device is yet but we need to get position data back from air brakes to figure out the next position to set them to. (I will have more info after my meeting with matt)

jjderooy, 2022-02-19:
The device is just a potentiometer. As you spin it the ADC it is connected to will send a reading of 0-1023 (most likely) to the flight computer GPIO pins. It's just like the VEX potentiometers we used in engr120 if you remember those.

jjderooy, 2022-04-07:
Hopefully mostly fixed in #53 . This at least lays the groundwork for it, though main will still need to actually put the data in rocketData from my understanding. None of the other data have integration with main yet so hopefully this is sufficient?

closed 2022-05-17 by klemie