UVicRocketry/Xenia1-Flight-Computer #39: X1-AV-25 Airbakes Stepper data

klemie, 2022-02-23:
**Description**
- @jjderooy 

**Acceptance Criteria**
 <Write specifically what is need for this issue to be completed>

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

jjderooy, 2022-02-26:
What do you mean by stepper data? They are open loop so there is no feedback. The feedback we get is from the encoder (potentiometer) which already has an issue: #30

jjderooy, 2022-02-26:
Ah gotcha. Stepper control is handled by the A4988s. Logan and I wrote some code for hybrid test stand steppers that can be used. I'll see if I can make a PR for it.

jjderooy, 2022-02-27:
@krislemie  Where should the code for controlling the stepper via a4988s go? It could go either in `main.py` or in `airbrakes.py`. The description for `airbrakes.py` is:
> Computes a the next value to set the airbrakes to position to. It does this through a kalman filter algorithm. Please look at the kalman filter algorithm connected to the issue X1-AV-6 

I could update this to include the physical control as well.

Lmk what you think and I'll make a PR.

klemie, 2022-02-27:
Everything should be decoupled from `main.py` so it should be in airbrakes.py a PR would be good

jjderooy, 2022-03-08:
Alright @klemie I'm done with my branch, but I can't test the code until the flight computer is assembled. Would you like a PR anyways?

closed 2022-03-18 by klemie