UVicRocketry/Xenia1-Flight-Computer #49: Convert Sub-Orbit from Matlab to Python

ebert-c, 2022-03-24:
**Description**
Sub-Orbit is a matlab program that will allow us to estimate our trajectory in real time.
We will be varying our drag utilizing the airbrakes to hit our target altitude.

We need to convert the Matlab program into Python and integrate it into the flight computer.
It will need to acquire sensor data and use that data to estimate trajectory.

**Acceptance Criteria**
The Sub-Orbit program is converted to Matlab and is able to receive sensor data from flight computer and output a trajectory.

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

PokeyOne, 2022-04-21:
https://github.com/UVicRocketry/Suborbit-Prediction-Script

PokeyOne, 2022-05-12:
Currently there are three main components of this that need to be done: The atmosphere function, the power phase, and the coast phase.

Currently the atmosphere component is done, the power phase is almost done, and should be done tonight. The coast phase will be very similar to the power phase so this is getting very close to the matlab script.

PokeyOne, 2022-05-12:
Next steps after this is done:
- Test on pi
- Is it fast enough?
- How are we integrating it with the rest?

PokeyOne, 2022-05-18:
Quick update on this for May 18 2022.

I think this is probably closable?

This issue seems to be mostly the conversion of the script.

The script is fully converted over at <https://github.com/UVicRocketry/Suborbit-Prediction-Script> , and most of the stuff left to complete is optimization and evaluation.

closed 2022-11-17 by PokeyOne