UVicRocketry/Xenia1-Flight-Computer #50: Issue: X1-AV-32 Control Stepper Motors Based on Needed Drag Coefficient

ebert-c, 2022-03-24:
**Description**
Given a needed drag coefficient, open the flaps controlled by the stepper motor to the position needed to achieve said drag.
A converter between drag coefficient and stepper motor position will be provided by Flight Dynamics.

**Acceptance Criteria**
When given a drag coefficient input, the system can cause the stepper motor to open the flaps to the position that provides that drag coefficient.

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

TechGal, 2022-06-09:
Follow up with claire.

TechGal, 2022-06-09:
Need updated drag coefficient list from claire.

closed 2022-11-17 by PokeyOne