UVicRocketry/Xenia1-Flight-Computer #70: X1-AV-34 Test Suborbit script on Pi

krislemie, 2022-05-12:
**Description**
Test the speed at which the simulation finish. Test the simulation on 1 thread with another tread running. 
**Acceptance Criteria**
 <Write specifically what is need for this issue to be completed>
- Speed at which the program runs on the pi is documented
- How many threads can be ran while program is running 
- Conclusions are made to weather this is a good option for predicting. 

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

jjderooy, 2022-05-17:
@PokeyOne can you give an update on where things are at with the optimization you've done?

PokeyOne, 2022-05-17:
The optimization I have done was in UVicRocketry/Suborbit-Prediction-Script#4

This took the execution time on the PI from 54s to about 0.4s

This was on the pi while @jjderooy was doing some very light testing on the pi, and the time does not include loading csv data and sorting the data. The load time shouldn't be included anyway though because it can be done once while the rocket is on the ground, and takes just a few seconds anyway.

The only other thing that I wanted to try off the top of my head was to make the `atmosphere` function inside the script just a LUT to see if that would be faster.

## Is this the right option?

There are a couple factors in whether this is the best option -- mainly accuracy and speed. As far as speed, I think it is reasonable ish. 0.4s ok. It will also take less time as it gets higher, because there is less to predict. I could also make it less precise, but I did already do that. I am hoping that this gets at least a little bit less as things go. Now, in terms of accuracy, it's a bit harder to tell. Currently the script is estimating roughly 12700 ft with full airbrakes and 14400 ft with no airbrakes. I do still need to get exact final mass data and the drag coefficient data will need to be updated with the latest changes to the fins. This is a little bit higher that what ASTOS is saying, but who knows really.

PokeyOne, 2022-05-17:
Also @klemie I updated some of the information in the issue if that is alright. And changed the formatting a little bit if that is alright. I was thinking it could make sense for the headings to be headings in issues, and for the acceptance criteria to be a check list.

closed 2022-11-17 by PokeyOne