UVicRocketry/Xenia1-Flight-Computer #74: X1-AV-38 Add HX711 to gpioReader

klemie, 2022-05-12:
**Description**
Use the driver (HX711Multi.py) to get 12 input strain readings put the function in gpioReader

**Acceptance Criteria**
- rocketData its updated with the strain data
- uploaded to pi and tested with testing rig in the rocket room ask @jjderooy or @krislemie how to set it up 

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
#56 

<Add to milestone>

jjderooy, 2022-05-16:
More context: create an instance of the HX711Multi class and pass it a GPIO pin for the clock, and a list of 12 GPIO pins for data. Then use the build in methods like tare(), isReady() and read() to read strain data from the sensors. There is lots of documentation at the top of the HX711Multi.py file.

krislemie, 2022-05-16:
> More context: create an instance of the HX711Multi class and pass it a GPIO pin for the clock, and a list of 12 GPIO pins for data. Then use the build in methods like tare(), isReady() and read() to [...]

Thanks JJ!