UVicRocketry/Xenia1-Flight-Computer #75: X1-AV-39 Pass Initial values into airbrakes

klemie, 2022-05-12:
**Description**
Initial values are needed inorder to create a prediction. 

**Acceptance Criteria**
 <Write specifically what is need for this issue to be completed>

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

jjderooy, 2022-05-17:
@PokeyOne is this required for suborbit as it is written currently?

PokeyOne, 2022-05-17:
@klemie What are initial values?

PokeyOne, 2022-05-17:
I think I kinda understand this. There will be some values that the script will need, such as launch altitude. During flight it will also need current altitude, velocity, and acceleration if available. If some of those values are not available directly, using some sort of estimate may be viable. There may be some things that could be done with a temperature and pressure reading on the ground, but that is not currently how it is written.

In summary:

Initial values:
- [ ] launch altitude

During flight
- [ ] altitude
- [ ] velocity
- [ ] acceleration

klemie, 2022-05-18:
I need to fill out this issue better