UVicRocketry/Xenia1-Flight-Computer #78: X1-AV-41 Send to transmitter

krislemie, 2022-05-16:
**Description**
Take the formatted data from RocketData.convert_to_csv_string and send that data to the transmitter using ___

**Acceptance Criteria**
 <Write specifically what is need for this issue to be completed>
- The transmitter receives the full package 

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

jjderooy, 2022-05-16:
Can you give some more context of how the transmitter connects? Name of transmitter, and libraries it uses that sort of thing? @TechGal any parameters we need to know about the antenna like character limits, packet frequency etc?

krislemie, 2022-05-16:
char limit is 200 but thats all I know

TechGal, 2022-05-17:
We're using the BigRedBee 70cm APRS [transmitter](https://shop.bigredbee.com/products/70cm-100mw-gps-aprs-transmitter). This transmitter uses a microcontroller which is initially programmed on a Windows computer, including a message which is sent to the ground with the automated GPS data and timestamp. The thing that needs to be determine is whether or not we can use the exposed UART interface to send a new message once / second (the recommended transmission speed). There are some programmers available, and I imagine there are some libraries too, for this microcontroller, but as this specific board is only used in a specific niche, I haven't found any tutorials explicitly discussing this board.

krislemie, 2022-05-19:
150c aprs package. Choose key values from rocketData to send.

krislemie, 2022-05-19:
we should figure out together in CDR

closed 2022-06-30 by macsyd