UVicRocketry/Xenia1-Flight-Computer #79: X1-AV-42 add Null checks

klemie, 2022-05-17:
**Description**
For all sensor, any missed value is `null`. Use the last saved value for calculations.

**Acceptance Criteria**
- null checks are added to all sensors read in gpioReader
- `convert_to_cvs_string` in rocketData is updated to send null in the string
- `send_all_data` in dataHandler  is updated to save null properly

**Dev Notes**
[for assignee]<write down your thought process when developing this issue>

**Linked issues**
 <If there are any related issues>

<Add to milestone>

jjderooy, 2022-05-17:
Are you sure writing null is the best option for the data logging? I'm just thinking for when we analyze the data afterwards with matplotlib or excel its gonna be a pain to remove all the nulls. Would it be better to just copy the previous value again instead of null?

PokeyOne, 2022-05-18:
I remember the discussion on this last week, and thinking about it more I think it may actually make more sense to log the previous value. 

I think the main argument for logging null would be to preserve the information that we weren't getting anything at that point, but this probably isn't all the important. If we look at the logs after and there is a streak of the exact same value to the 5th decimal place for half the flight we'll be able to tell if nothing was changing. It would also just simplify all the logic and reduce errors within the program. If we just have one variable that stores the previous value then there is one source of truth that is never null. If we start messing with null then we are going to have to check for null on every singe equation and it will get messy real quick.

Also it would just make things easier like JJ said.

Another option would be to just put an empty cell. I know people were saying would make it harder to read last week, but I don't think the CSV is really meant to be read by a human anyway. Most of the time if someone is looking at the data it should just be in some sort of software like excel. I still think I prefer having the previous value over an empty cell, but personally I think an empty cell is better than just writing `null`

Writing null can also be a bit misleading, because it makes it seem like we don't have data. We do have data. It just hasn't been updated that particular cycle.

j7uu, 2022-06-09:
Notes/Discussion From Avionics Meet (2022-06-08)
- Could use last value of the sensor for x amount of time (then enter timeout and use generic value)
- Discussed the merits and pitfalls of using old flight data for generic values (values based on time, values from comparing current sensors to old data, etc)
- Try catch all of main????????