UVicRocketry/Xenia1-Flight-Computer #85: X1-AV-53 Change stepper driver for airbrakes

klemie, 2022-05-24:
# Description
 <Describe what the feature or bug that need to be resolved>
 <Description of the big component that encompasses the issue working on>
 <Depencies, ex. libraries, sensors, hardware, small parts of code youhave to work on>
 <ow to do the things>
 NOTE: if you import anything new add it to the list in `requirements.txt`
Spec new more reliable stepper driver for airbrakes. @Devon-H knows the right one
# Acceptance Criteria
 <Write specifically what is need for this issue to be completed>

- [ ] Integrate new sensor in cad 
- [ ] Learn how to control it with python
- [ ] Write new ticket for how to update airbrakes.py on how to read it

# Dev Notes
[for assignee]<write down your thought process when developing this issue>

# Linked issues
 <If there are any related issues>

<Add to milestone>

Devon-H, 2022-05-24:
The stepper motor driver in question is the DRV8825. Here's an amazon [link](https://www.amazon.ca/Baluue-Stepper-Compatible-StepStick-Printer/dp/B08PV6JTK7/ref=sr_1_43?crid=9692R8CDHXYP&keywords=stepper+motor+driver&qid=1653415285&sprefix=stepper+motor+drive%2Caps%2C209&sr=8-43) for reference

PokeyOne, 2022-06-29:
I just got assigned to this, but this may or may not already be finished. I know for a fact that I already updated the schematic, but I haven't looked at the code. I'll take a look later today.

If it is already updated, I will still want to test that it works before closing this issue.

closed 2022-11-17 by PokeyOne