UVicRocketry/Xenia1-Flight-Computer #89: X1-AV-57 Update main.py

krislemie, 2022-05-25:
# Description
 <Describe what the feature or bug that need to be resolved>
 <Description of the big component that encompasses the issue working on>
 <Depencies, ex. libraries, sensors, hardware, small parts of code youhave to work on>
 <ow to do the things>
 NOTE: if you import anything new add it to the list in `requirements.txt`

Follow the sequence diagram in the readme. Make sure to break the logic up into the states as shown in the readme. 

1. Setup Phase
- Waiting for input from screw switch (powered off - no code needed)
2. Standby Phase
- Initialize sensors, load suboribit csv, sort csv(will be done in suborbit). 
- Then in a loop call read necessary data for
3. Powered Flight Phase

4. Coast Phase

5. Recovery

# Acceptance Criteria
 <Write specifically what is need for this issue to be completed>
Code is written has all these phases
- [ ] Setup up phase - waiting for input from screw switch (powered off - no code needed)
- [ ] Standby Phase - initialize sensors, load suboribit csv, sort csv. Then in a 
- [ ]  

# Dev Notes
[for assignee]<write down your thought process when developing this issue>

# Linked issues
 <If there are any related issues>

<Add to milestone>

PokeyOne, 2022-05-25:
These are the states of main as I understand them:

```mermaid
stateDiagram-v2
  setup   : Setup
  standby : Standby
  abort   : Abort
  power   : Powered Flight
  coast   : Coast
  reco    : Falling + Recovery
  [*] --> setup     : Turn key
  setup --> standby : All setup tasks are complete
  standby --> abort : System manually turned off
  standby --> power : Ignition detection
  power --> coast   : Burnout detection
  coast --> reco    : Apogee detection
  reco --> [*]      : Manual power off
```

**Note**: Falling and recovery are the same state because at that point it doesn't really matter to flight computer. The flaps should be retracted in both.

PokeyOne, 2022-05-25:
Also @klemie, was just thinking, maybe this issue could be narrowed down to just setting up the main to have a state machine, and then the actual states can be broken up into separate issues. Once things are setup to have states, they can be fairly isolated from each other. This way we can have multiple people working at the same time.

this is a bit of a rough idea that I had:

```mermaid
gantt
  title Main State Machine
  dateFormat YYYY-MM-DD
  excludes weekends
  section Milestones

  Assembly Rehearsal           :milestone, assembly, 2022-06-04, 0d
  Second Assembly Rehearsal    :milestone, aseembly2, 2022-06-11, 0d
  System Acceptance Review     :milestone, acceptance, 2022-06-25, 0d

  section Setup

  Setup state machine logic    :active, init, 2022-05-25, 7d

  section Dependencies

  Ignition detection           :        ignition, 2022-05-25, 7d
  Accel and vel calculations   :        accelvel, 2022-05-25, 7d
  Motor burnout detection      :        burnout, after accelvel, 5d
  Apogee detection             :        apogee, after burnout, 5d

  section States

  Setup                        :        setup, 2022-06-01, 2d
  Standby                      :        standby, after init, 4d
  Powered flight               :        powered, after init, 4d
  Coast flight                 :        coasting, after init, 7d
  Recovery                     :        falling, after standby, 2d
```

Got kinda excited about Gantt charts after the ECE 299 lecture today. Some of the tasks will probably need to be adjusted; this was just an idea that I had and ran with it.

Also these charts are just Mermaid JS charts written as text in the GitHub comment, but I could put these somewhere more permanent at some point.

PokeyOne, 2022-05-26:
I put this into in-progress because PR #99 does at least some of this.

closed 2022-06-09 by TechGal