UVicRocketry/Xenia1-Flight-Computer #108: Update All Files to Use `snake_case` for Variables and Methods.

PokeyOne, 2022-06-05:
# Description

There are many places where we will used `snake_case` in one file and then in another file we will use `lowerCamelCase` and `UpperCamelCase`. We should be just using `snake_case` for all functions and variables. This does not apply to classes.

One example of this is in the `gpioReader.py` file. This should be `gpio_reader.py` and inside all methods should be transformed to snake case as well.

# Acceptance Criteria

- [ ] All variables are snake case
- [ ] All functions are snake case
- [ ] All file names are snake case

# Dev Notes

[for assignee]<write down your thought process when developing this issue>

krislemie, 2022-07-15:
can you pull in the changes from refactor and update accordingly?

PokeyOne, 2022-07-15:
@krislemie Yes, but after assembly rehearsal I think is a better time. This is just a style thing and isn't strictly necessary right now. After assembly rehearsal I'll update the PR and then we can merge it.

closed 2022-11-17 by PokeyOne